tscircuit/schematic-viewer #48: `<board />` center alignment bug

imrishabh18, 2024-07-20:
File - https://github.com/tscircuit/schematic-viewer/blob/714e6a0189cd7ba887ffb20a4add964ba086417e/src/stories/bugs/bug8-autolayout.stories.tsx

`<board />` props `pcbCenterX` and `pcbCenterY` which are depreciated when replaced with `pcbX` and `pcbY` is having bug

| Current    | After replaced with new props val  |
| -------- | ------- |
| [image]  | 
  
[image] |

seveibar, 2024-07-20:
Anything in the console?

imrishabh18, 2024-07-21:
The props need to be updated, it seems.

[image]

seveibar, 2024-07-21:
I just patched this in the latest version of builder yesterday https://github.com/tscircuit/builder/pull/79

cc @Slaviiiii

closed 2024-08-03 by imrishabh18